mcci-catena/arduino-lorawan #25: Finish implementing session save/restore

terrillmoore, 2018-06-02:
While studying #21, I discovered that I had forgotten to implement session restore. Just as well, as the session format I chose didn't represent all the needed information. But this needs to be finished.

ThomasBlt, 2020-01-03:
@terrillmoore hasn't this been dealt with by the addition of [GetSavedSessionInfo](https://github.com/mcci-catena/arduino-lorawan/blob/703e776f511d93733a3f294fe0fa4641e294e3db/src/Arduino_LoRaWAN.h#L556) ?
If true what's the need for [NetSaveFCntUp](https://github.com/mcci-catena/arduino-lorawan/blob/703e776f511d93733a3f294fe0fa4641e294e3db/src/Arduino_LoRaWAN.h#L594) and [NetSaveFCntDown](https://github.com/mcci-catena/arduino-lorawan/blob/703e776f511d93733a3f294fe0fa4641e294e3db/src/Arduino_LoRaWAN.h#L604) as there is no restore function for them ?

jangrewe, 2020-07-31:
Yes, please! I've made the first steps with LoRa on an ESP32, and this wrapper helped me a lot!
I've got most functionality working, now all i need is some way to save/restore the session info - but my coding skills are lacking, a lot, to be able to implement this.

jdeus, 2020-09-29:
Still open ?

jangrewe, 2020-09-29:
@jdeus have a look right below the title of this issue. if you see a green badge that says "(!) Open", then yes, this issue is still open.

commit 6158a1c8 (2021-05-06): Fix #25: implement complete session save/restore
closed 2021-05-06 by terrillmoore