mcci-catena/arduino-lorawan #40: Question on RX data

trlafleur, 2018-07-06:
If I send data to the device and I ask for a confirmation that the data was received using the LiMC stack, how do I send the ACK back to TTN??

I can see the data in the device, but TTN keeps sending it?? ie: it's waiting for an ACK...

Thanks

terrillmoore, 2019-04-01:
@trlafleur are you still seeing this? Sorry, because I didn't immediately assign this to myself, it got lost.

trlafleur, 2019-04-01:
I believe so, will need to retest

trlafleur, 2019-06-01:
Terry:  I noticed in the current version 2.3.1 of your LMiC code that you
have an example:  ttn-otaa-network-time

I take it needs some additional code at the server to function?? any
reference to it??

Thanks

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

terrillmoore, 2019-07-06:
Sorry @trlafleur -- just saw this.

- TTN doesn't yet support network-time.
- LoRaServer.io does, I think.
- It's possible to do something similar at a higher level with TTN (in your network app). If you check the conversation around the network time fix, you should be able to find someone who did this in Switzerland succesfully, with sample code.  (You take the tagged "time of arrival", and send it back to the device at the application layer.)