mcci-catena/arduino-lorawan #49: Please fix example program in README.md

petesoper, 2018-11-09:
Has publc"  instead of "public".
    The two class constructors don't have the same name as the class.
    The code for the second constructor is outside the curly braces and has no semicolon.
    But #3 and some other errors I won't mention are a side effect of the fact that "limic_pinmap" is not defined. Could you give us a clue about what to include to satisfy this? There may be other problems, but I decided to stop with the undefined structure. (it is defined in many places in the libraries)
    The build was attempted with Arduino 1.8.7 and these three libraries installed:
  https://github.com/mcci-catena/arduino-lorawan
  https://github.com/mcci-catena/arduino-lmic
  https://github.com/mcci-catena/Catena-mcciadk
Target board is an "Adafruit Feather M0" installed from here:
https://adafruit.github.io/arduino-board-index/package_adafruit_index.json

Thanks

terrillmoore, 2018-11-09:
Thanks for your comments. Sorry for the problems. I'll look into that this weekend.

--Terry

petesoper, 2018-11-11:
To clarify the reference to "#3" above, I had mention of a limic_pinmap parameter mention shadowing a reference in the body of the second constructor and decided that wasn't important. So "#3" was about the shadowing that was just an artifact of the lack of a definition for limic_pinmap.
Thanks for jumping on this!

terrillmoore, 2018-11-12:
hi @petesoper -- i improved the readme, but I am working on an example. this interacts with a release that I'm doing for arduino-lmic, so it didn't get posted yesterday. I hope to get it done today.

commit 0ba36ff6 (2018-11-12): Merge pull request #52 from mcci-catena/issue49

Fix #49: add compilable example, update README, add to travis
commit 9bbd0878 (2018-11-12): Fix #49: add compilable example, update README, add to travis
closed 2018-11-12 by terrillmoore