mcci-catena/arduino-lorawan #63: TTN header files getting included despite #include of the machineQ header

chipmc, 2018-11-28:
First, let me thank you for the excellent work you are doing with these LMIC and LoraWan libraries.  

In my sketch, I am only including the following:

#include <Arduino_LoRaWAN_machineQ.h>

Yet the sketch fails to compile with the error below.  Somehow, the Arduino_LoRaWAN_ttn header is getting included and looking for a preconditions file that does not exist.  I believe I can comment out this code as I am not using PlatformIO but, I want to use the MachineQ service and don't want to use ttn_as923_netbegin.cpp so, I would like to know what I need to do to ensure the proper net begin is being used.  

MY sketch worked properly until the Arduino helpfully updated the MCCI-catena/arduino-limic yesterday so I assume this is an issue of compatibility between these libraries.

Thanks, Chip

Arduino: 1.8.5 (Mac OS X), Board: "Adafruit Feather M0"

In file included from /Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/lib/ttn_as923_netbegin.cpp:34:0:
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN_ttn.h:53:67: fatal error: ../../arduino-lmic/src/lmic/lmic_config_preconditions.h: No such file or directory
 #include "../../arduino-lmic/src/lmic/lmic_config_preconditions.h"
                                                                   ^
compilation terminated.

terrillmoore, 2018-11-28:
Hm. Looks like the two libraries are out of sync, and you have an *older* version of arduino-lmic. Can you check the version? Is it possible that you had a manual install and they downgraded you?  (I rather suspect that is what happened.)  We pushed the 2.3.0 release and it definitely has the "missing" file.  Alternately, they put the library someplace else (something that isn't anticipated by that design, as you might guess), rather than in your Arduino/libraries directory?

terrillmoore, 2018-11-28:
@sjpark-mcci can you try doing a library-manager install of the lmic, and a manual install of the arduino-lorawan, and see if perhaps this problem crops up for you? I rather suspect that this is the problem.

@chipmc if this is the problem, I know what to do; we'll have to push a quick release to patch this. Sorry, obviously this isn't part of our test matrix (manual + automatic install). In any case, I now realize that this is a valid scenario and it need to be corrected.

chipmc, 2018-11-28:
OK, I have the following libraries installed:
Catena MCCI Arduino Development Kit - v0.1.3
MCCI Arduino LoRaWAN Library - v0.4.0
MCCI LoRaWAN LMIC Library - v2.3.0

Should I get rid of the LoRaWAN v0.4.0 library?

chipmc, 2018-11-28:
OK, so I did a manual uninstall and reinstall of MCCI Arduino LoRaWAN Library - so it is now v0.5.0.

I also noticed that when the Arduino Library manager installed the MCCI LoRaWAN LMIC Library, it left the name of the folder as just that.  However, in the error message, I saw it assumes that the directory needs to be Arduino-lmic.  I made that change and was rewarded with an entirely different set of error messages.  This feels like progress but any help you could provide on this latest crop would be appreciated.  

Thank you, Chip

****************************************************************

Arduino: 1.8.5 (Mac OS X), Board: "Adafruit Feather M0"

In file included from /Users/chipmc/Documents/Personal/Maker/Arduino/ttn-otaa-feather-us915/ttn-otaa-feather-us915.ino:34:0:
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN_machineQ.h: In constructor 'Arduino_LoRaWAN_machineQ_base::Arduino_LoRaWAN_machineQ_base(const lmic_pinmap&)':
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN_machineQ.h:64:81: error: no matching function for call to 'Arduino_LoRaWAN::Arduino_LoRaWAN(const lmic_pinmap&)'
         Arduino_LoRaWAN_machineQ_base(const lmic_pinmap & pinmap) : Super(pinmap) {};
                                                                                 ^
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN_machineQ.h:64:81: note: candidates are:
In file included from /Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN_machineQ.h:43:0,
                 from /Users/chipmc/Documents/Personal/Maker/Arduino/ttn-otaa-feather-us915/ttn-otaa-feather-us915.ino:34:
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN.h:299:2: note: Arduino_LoRaWAN::Arduino_LoRaWAN()
  Arduino_LoRaWAN();
  ^
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN.h:299:2: note:   candidate expects 0 arguments, 1 provided
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN.h:51:7: note: constexpr Arduino_LoRaWAN::Arduino_LoRaWAN(const Arduino_LoRaWAN&)
 class Arduino_LoRaWAN
       ^
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN.h:51:7: note:   no known conversion for argument 1 from 'const lmic_pinmap {aka const Arduino_LMIC::HalPinmap_t}' to 'const Arduino_LoRaWAN&'
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN.h:51:7: note: constexpr Arduino_LoRaWAN::Arduino_LoRaWAN(Arduino_LoRaWAN&&)
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN.h:51:7: note:   no known conversion for argument 1 from 'const lmic_pinmap {aka const Arduino_LMIC::HalPinmap_t}' to 'Arduino_LoRaWAN&&'
exit status 1
Error compiling for board Adafruit Feather M0.

terrillmoore, 2018-11-29:
yes, the name of the library is certainly a problem; I'll try to push a fix now.. However I can't repro this problem, and it looks like a problem I had at an intermediate state (when a file had not been updated). Could you try renaming your "libraries" directory (or pointing at a new empty sketches dir) and just install the three libraries from scratch again? I suspect that something got borked in the struggle til now.

terrillmoore, 2018-11-29:
Header file issue closed by #65. If you update to v2.3.1 and v0.5.1, respectively, they should build without editing the header file or renaming the libs.

closed 2018-11-29 by terrillmoore

terrillmoore, 2018-11-29:
@chipmc wrote

> Thanks for your help

Thanks in turn for your patience in working through this.
--Terry