mcci-catena/arduino-lorawan #64: LoRaWAN fails header test

chipmc, 2018-11-28:
I thought I would simplify the issue and start a new thread.  I hope this is OK.  

I was able to fix the last issue but I can't get my sketch to compile.  To take my sketch out of the discussion, I thought I would try to compile the header test sketches from the example folder.  

I have pasted the error messages I get when I attempt to compile header_test_ttn.ino unmodified from the examples folder below.  

Here are the libraries I am using:
Catena MCCI Arduino Development Kit - v0.1.3
MCCI Arduino LoRaWAN Library - v0.5.0 - manual install
MCCI LoRaWAN LMIC Library - v2.3.0 - Arduino Library Manager install - renamed folder to arduino-lmic

Any help would be appreciated.

Thanks, Chip

*************
Arduino: 1.8.5 (Mac OS X), Board: "Adafruit Feather M0"

In file included from /Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/lib/machineq_base_netbegin.cpp:34:0:
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN_machineQ.h: In constructor 'Arduino_LoRaWAN_machineQ_base::Arduino_LoRaWAN_machineQ_base(const lmic_pinmap&)':
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN_machineQ.h:64:81: error: no matching function for call to 'Arduino_LoRaWAN::Arduino_LoRaWAN(const lmic_pinmap&)'
         Arduino_LoRaWAN_machineQ_base(const lmic_pinmap & pinmap) : Super(pinmap) {};
                                                                                 ^
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN_machineQ.h:64:81: note: candidates are:
In file included from /Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN_machineQ.h:43:0,
                 from /Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/lib/machineq_base_netbegin.cpp:34:
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN.h:299:2: note: Arduino_LoRaWAN::Arduino_LoRaWAN()
  Arduino_LoRaWAN();
  ^
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN.h:299:2: note:   candidate expects 0 arguments, 1 provided
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN.h:51:7: note: constexpr Arduino_LoRaWAN::Arduino_LoRaWAN(const Arduino_LoRaWAN&)
 class Arduino_LoRaWAN
       ^
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN.h:51:7: note:   no known conversion for argument 1 from 'const lmic_pinmap {aka const Arduino_LMIC::HalPinmap_t}' to 'const Arduino_LoRaWAN&'
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN.h:51:7: note: constexpr Arduino_LoRaWAN::Arduino_LoRaWAN(Arduino_LoRaWAN&&)
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN.h:51:7: note:   no known conversion for argument 1 from 'const lmic_pinmap {aka const Arduino_LMIC::HalPinmap_t}' to 'Arduino_LoRaWAN&&'
exit status 1
Error compiling for board Adafruit Feather M0.

chipmc, 2018-11-29:
OK, wow, not kidding when they said verbose!.  I have attached the file.

Thank you for your help, Chip

[Verbose Output MCCI LoRaWAN TTN sketch errors.txt](https://github.com/mcci-catena/arduino-lorawan/files/2627107/Verbose.Output.MCCI.LoRaWAN.TTN.sketch.errors.txt)

closed 2018-11-29 by chipmc

chipmc, 2018-11-29:
Hmmm, looking at the output I am a bit confused.  

The sketch is the one you provided in your examples - no modification.  

Yet, when I look at the verbose output, I see it is referencing the machineQ header file here:

In file included from /Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/lib/machineq_base_netbegin.cpp:34:0:
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN_machineQ.h: In constructor 'Arduino_LoRaWAN_machineQ_base::Arduino_LoRaWAN_machineQ_base(const lmic_pinmap&)':
/Users/chipmc/Documents/Personal/Maker/Arduino/libraries/arduino-lorawan/src/Arduino_LoRaWAN_machineQ.h:64:81: error: no matching function for call to 'Arduino_LoRaWAN::Arduino_LoRaWAN(const lmic_pinmap&)'
         Arduino_LoRaWAN_machineQ_base(const lmic_pinmap & pinmap) : Super(pinmap) {};

How is this possible?  I do want to use your library for machineQ but, I figured the best way to isolate the issue was to use the example sketches you provided, and there is only ttn examples.  

why is this code touching Arduino_LoRaWAN_machineQ ?

Thank you again for your help.  As frustrating as these compile errors are, I end up learning something which generally makes it all worthwhile.

Chip

terrillmoore, 2018-11-29:
This also looks like you have an ancient version lurking. I'm goign to open a new bug for the header problem, (../../arduino-lmic/...) because that's a real problem.

terrillmoore, 2018-11-29:
The reason the machineQ header file is being included is because the machineQ .cpp files are always *compiled*; we depend on the linker to take them out of the final image if not used. Those files include the machineQ headers.. It makes certain tasks less painful. As long as you don't have `ARDUINO_LORAWAN_NETWORK_MACHINEQ` defined, the code is compiled but not used.

chipmc, 2018-11-29:
Terry, 

That makes sense.  I do plan to use this library for machineQ so, that is the next step.  I next plant to understand what the machineQ customizations are and use them.  Thanks again for your help. 

Chip