mcci-catena/arduino-lorawan #101: Library does not recognize the Adafruit Feather M0 board correctly.

chipmc, 2019-04-16:
On compile and when running the compliance script, I get the following message: "board not known to library; add pinmap or update getconfig_thisboard.cpp".

If I change the board to the MCI Catena 4410, the compliance script compiles and runs without this error.

paulalting, 2019-04-16:
A quick and maybe dirty fix I did is as follows, which allows you to then define the board in the user config file named 'lmic_project_config.h' which resides in folder 'project_config'

In file 'src/arduino_lmic_hal_boards.h' add #include at line 22 after initial #include.
`#include "lmic/lmic_config_preconditions.h"`

Terry, does the standard Arduino IDE define this board name by default or does it need to be set by user as the Eclipse IDE does not make such a definition as ARDUINO_SAMD_FEATHER_M0 ?

terrillmoore, 2019-04-16:
Try the feather m0 express as your target. I just checked boards.txt from the latest Adafruit BSP and they define as follows:

Feather M0
-D ARDUINO_SAMD_ZERO -D ADAFRUIT_FEATHER_M0

Feather M0 Express
-D ARDUINO_SAMD_ZERO -D ARDUINO_SAMD_FEATHER_M0 ....

Evidently I was looking at the wrong line in the BSP, or I used the Feather M0 Express, or something. I'll update our #ifdefs to accommodate. Or you can patch the #ifdef to look for ADAFRUIT_FEATHER_M0.

chipmc, 2019-04-16:
Selecting the Adafruit M0 Feather Express fixes the issue with board not being recognized.  Thank you.

terrillmoore, 2019-04-16:
Good. I'll fix this in the next day or two. And maybe add a CI test.

paulalting, 2019-04-16:
Yes, same here, if I select the Feather M0 Express in the project properties in Eclipse IDE and remove the dirty fix I had above, all compiles nicely now.