mcci-catena/arduino-lorawan #126: Can not compile master

schenlap, 2019-12-19:
I get the following error
`WARNING: library brzo_i2c-master claims to run on esp8266 architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
In file included from /home/xxx/Arduino/libraries/arduino-lorawan-master/src/lib/GetRegionString.cpp:19:0:
/home/xxx/Arduino/libraries/Catena-mcciadk-0.2.1/src/mcciadk_baselib.h: In function 'size_t McciAdkLib_BufferToUint32(const char*, size_t, unsigned int, uint32_t*, bool*)':
/home/xxx/Arduino/libraries/Catena-mcciadk-0.2.1/src/mcciadk_baselib.h:49:64: error: invalid conversion from 'uint32_t* {aka unsigned int*}' to 'long unsigned int*' [-fpermissive]
  return McciAdkLib_BufferToUlong(s, n, base, pulnum, pfOverflow);
                                                                ^
/home/xxx/Arduino/libraries/Catena-mcciadk-0.2.1/src/mcciadk_baselib.h:31:1: note:   initializing argument 4 of 'size_t McciAdkLib_BufferToUlong(const char*, size_t, unsigned int, long unsigned int*, bool*)'
 McciAdkLib_BufferToUlong(
 ^
In file included from /home/xxx/Arduino/libraries/arduino-lorawan-master/src/lib/arduino_lorawan_begin.cpp:20:0:
/home/xxx/Arduino/libraries/Catena-mcciadk-0.2.1/src/mcciadk_baselib.h: In function 'size_t McciAdkLib_BufferToUint32(const char*, size_t, unsigned int, uint32_t*, bool*)':
/home/xxx/Arduino/libraries/Catena-mcciadk-0.2.1/src/mcciadk_baselib.h:49:64: error: invalid conversion from 'uint32_t* {aka unsigned int*}' to 'long unsigned int*' [-fpermissive]
  return McciAdkLib_BufferToUlong(s, n, base, pulnum, pfOverflow);
                                                                ^
/home/xxx/Arduino/libraries/Catena-mcciadk-0.2.1/src/mcciadk_baselib.h:31:1: note:   initializing argument 4 of 'size_t McciAdkLib_BufferToUlong(const char*, size_t, unsigned int, long unsigned int*, bool*)'
 McciAdkLib_BufferToUlong(`

The documentations says that Catena-mcciadk 0.1.1 is needed, but on https://github.com/mcci-catena/Catena-mcciadk/releases there is no 0.1.1. Comiling does not work with 0.1.3 nor with master of Catena-mcciadk.

terrillmoore, 2019-12-19:
Sorry you're having problems.  This is a compatibility problem; we haven't even test-compiled this for ESP32. From the compile log, this is a `uint32_t` != `ulong` error. Indeed, the line before the error says `// this will throw a compile error if uint32_t != ulong`.  It is not clear to me that compilers should fail a compile if sizeof(uint32_t) == sizeof(ulong), but you can't fight city hall. You could edit that function as:

```c++
#if defined(ARDUINO_ARCH_ESP32)  // you'll have to confirm this symbol
	unsigned long nonce;
	size_t const result = McciAdkLib_BufferToUlong(s, n, base, pulnum ? &nonce : NULL, pfOverflow);
	if (pulnum)
	    *pulnum = nonce;
	return result;
#else
	// this will throw a compile error if uint32_t != ulong
	return McciAdkLib_BufferToUlong(s, n, base, pulnum, pfOverflow);
#endif
```

The extra complexity is exactly why I didn't make it this way by default. In C, unlike C++, there's no good way to see whether uint32_t is different than ulong; and I don't want to burden AVR, small ARM, etc, with this extra code.

terrillmoore, 2019-12-19:
Sorry, closed by accident. However, I will transfer this to the ADK when I fix it, because this is not actually a bug in the arduino-lorawan library.

schenlap, 2019-12-20:
Compiles now, thank you