mcci-catena/arduino-lorawan #143: Fully functioning example

kiranshila, 2020-06-17:
I'm still quite confused on some of the moving parts in this library, especially with regards to what I *don't* call from LMIC. For example, do I implement `doEvent`? Or do I use `RegisterListener`?

It would be extremely useful to see an entire example using this library to just send a Hello World, other than just verifying that the lib compiles.

andrewwootton, 2020-07-19:
yes I agree.  I've used the original lmic library before with success but this new implementation allows a much improved program structure - accolades for doing this.
But I'm struggling to understand how to provision the feather example with my deveui appeui etc and how to set the region and network.  Does this happen in my program or in a config file?  In the older MCCI_LoRaWAN_LMIC_library there was a project_config file where the region was set, but the code was specified the deveui etc.
Sorry for my lamentable knowledge.
Andrew

terrillmoore, 2020-07-19:
Hi all, sorry -- I use this library with Catena-Arduino-Platform, and so I've not really needed an independent test case (all my tests are over there). I've been contemplating merging this into the LMIC. One thing to be aware of is that in the LMIC, I'm working on secure element support with a couple of my colleagues (see branch [devel-issue578](https://github.com/mcci-catena/arduino-lmic/tree/devel-issue578)), and this necessarily changes how keys work... a lot....  We hope to have that finished by the end of July, but of course that depends on my time availability.

jangrewe, 2020-07-31:
While not a **fully** working example (e.g. no session save/restore), i got at least something working to offer:
https://git.faked.org/jan/bikebeam/-/tree/master
The only interesting files are `platformio.ini` and the contents of `src`.

It works in terms of "my TTGO T-Beam (an ESP32) sends messages to TTN" - but it still rejoins the network on each reset, as i haven't figured out how to save/restore the session info.

pkerspe, 2021-03-21:
> While not a **fully** working example (e.g. no session save/restore), i got at least something working to offer:
> https://git.faked.org/jan/bikebeam/-/tree/master
> The only interesting files are `platformio.ini` and the contents of `src`.

The platformio.ini build flags saved my day! Would be helpful to point these settings out in the documentation for setting up the project in platformio

avlbger, 2021-05-16:
I also really need an example that shows how one can use this library. I was able to get the mcci-catena lmic up and running but how can i use this arduino-lorawan now on top?
The included examples just simply do nothing or am i  missing something? Searched a lot but did not find much, managed to add a pinmapping but stuck at trying to register a eventlistener so i could see whats going on.
Does anyone has a basic example how to use and configure this lib? (or how exactly this works together with the mcci lmic-lib)

terrillmoore, 2021-05-17:
Sorry! MCCI uses this library in conjunction with [Catena-Arduino-Platform](https://mcci-catena/Catena-Arduino-Platform).  There are extra layers of wrappers because we integrate our FRAM support in that library.  Unluckily, that is *too much* extra; those examples also make it hard to figure out what's going on.

One clue: using this library, you don't use the LMIC event listeners; the Arduino-LoRaWAN provides an event listener to the LMIC, and you register with the Arduino-LoRaWAN object. It translates all the events into outcalls on various registered APIs. If you use something like Visual Studio Code and trace through examples in that library, it should become pretty obvious. You will have to install the MCCI BSP for STM32 for best results; but that just lets you get Intellicode working completely.

There is nothing special about this library (unlike the LMIC) -- its the same as any other Arduino library and requires no special treatment from PlatformIO. The LMIC is a different story, of course. We don't use PlatformIO at MCCI; our target users are using Arduino, and we can't maintain multiple BSPs; but I'm sure if you temporarily use VS Code + Arduino as a tool for understanding, it will become obvious what to do.

fsalomon, 2021-05-17:
Should I state the obvious? Arduino people are not C++ pros.

I understand that you @terrillmoore may not be able to commit time to providing a working example but the lack of documentation on this library is definitely a blocker for many.

Aren't there any other projects apart from Catena using parts of this library? Can't we get a sponsor (I'm looking at you @helium) on board to provide better documentation?

terrillmoore, 2021-05-17:
Hi @fsalomon - you may find concrete examples if you look through the directories of people who've forked the repo. I know @frankleonrose used a slightly earlier version of this in one of his projects.

As it happens, not many people seem to be using the MCCI libraries on non-MCCI hardware; and due to the shortages / nonexistence of the Murata module, we can't sell much in the way of MCCI hardware. I'm pushed to prioritize things for projects that are focused on specific customers to maximize revenue from limited supply. I suppose I could put out a @helium tip jar; we have a gateway but we're certainly not making anything remotely like the amount that others report, nothing that supports investment.

After I get MCCI's trusted bootloader published for STM32L0 (link will be https://github.com/mcci-catena/bootloader), I may be able to put a little attention on an example. Next maybe later this month.

fsalomon, 2021-05-17:
Well I think there is causality here: Not many people using it because it's hardly usable. My impression is that this library is still the best bet to do some basic stuff on cheapo hardware -- which is what Arduino is all about imho.

For example my hardware shipped with https://github.com/cyberman54/ESP32-Paxcounter which is built on https://github.com/mcci-catena/arduino-lmic but the project is so convoluted that stripping it off the non-LoRaWan stuff would be a nightmare.

For anyone interested, this seems to be the code @terrillmoore mentioned:
https://github.com/frankleonrose/LoraStack/blob/master/src/LoraStack.cpp
https://github.com/frankleonrose/LoraStack/blob/master/examples/LoraStackBasic/LoraStackBasic.cpp

avlbger, 2021-05-17:
@terrillmoore thanks for your answer and the hint to search through the forks of the lib to may find missing details was helpful i guess :) at least i could have thought of that myself... will dig in there and hopefully find all things to get it working... i am not a person that is too lazy to find out things myself but i really felt a little lost yesterday when i spent many hours with no result ;)

@fsalomon thanks also to you and for providing the link of some may helpful code

terrillmoore, 2021-05-17:
@fsalomon there may be causality, but our use case for this library (the Catena code) works very well with this use, and it's used in that context all the time. I'd be happy to have contributions and to support contributions by answering questions. If I have time, I'll do my best to improve things.

@avlbger you might also find @lnlp's https://github.com/lnlp/LMIC-node project helpful in terms of how it uses the LMIC directly. @lnlp is very careful and supports lots of different platforms. (And he asks me lots of questions at [forum.mcci.io](https://forum.mcci.io) -- the discussion there might be informative.)

fsalomon, 2021-05-17:
So I suggest we gather some more examples here and then create 
https://github.com/mcci-catena/arduino-lorawan/tree/master/examples/README.md and 
https://github.com/mcci-catena/arduino-lmic/tree/master/examples/README.md
and interlink the two so that people can hit the ground running when they find those libraries.

terrillmoore, 2021-05-31:
Help on documentation gratefully accepted. As it happens, for debugging purposes I just created an app that may serve as a suitable starting point. See branch issue143 (in process)

commit e28a9f81 (2021-06-01): Fix #143: add complete example
closed 2021-06-01 by terrillmoore