mcci-catena/arduino-lorawan #144: How to implement low power LoRaWAN device using OTAA?

wzab, 2020-09-17:
I need to implement a LoRaWAN device that sends a message every 10 minutes. To limit the power consumption I need to put the device to sleep between sending the messages.
If I use ABP, the problem can be solved easily with  simple putting the CPU to sleep in the loop() function.
However such solution is not possible when using OTAA, as it breaks the timing,
I have checked the sources, and I can see that Arduino_LoRaWAN::loop() https://github.com/mcci-catena/arduino-lorawan/blob/4bc0d48b9956d5bed8e26dda01ead05d8ac269b3/src/lib/arduino_lorawan_loop.cpp#L40 calls the os_runloop_once routine from the underlying Arduino-LMIC library, and this routine [calls](https://github.com/mcci-catena/arduino-lmic/blob/6fe04ec76348c1c3ec808d5eb5e41ddad477591c/src/lmic/oslmic.c#L140) hal_sleep() .
Unfortunately, the [hal_sleep()](https://github.com/mcci-catena/arduino-lmic/blob/6fe04ec76348c1c3ec808d5eb5e41ddad477591c/src/hal/hal.cpp#L368) function is currently not implemented .
Does it mean that if I need to implement a low-power LoRaWAN device I have to use ABP or I need to use another, not-MCCI LoraWAN library?

wzab, 2020-09-18:
I have found  [that website](https://mariozwiers.de/2018/04/04/arduino-lmic-library-with-low-power-mode/). It requires me to use the low-level library, but seems to support the deep sleep mode.

jdeus, 2020-09-29:
Yes these LMIC libraries are for from being usable. They lack key features such as saving the state to be restored when exiting a deep sleep.

Much work has to be done for this to be used in any serious application.

postfalk, 2020-11-02:
It says "It provides a simple framework for doing low-power programming." in the documentation and never gets back to the topic. Documentation needs to be improved. This library is not user-friendly. See also #143