mcci-catena/arduino-lorawan #146: m_nRegisteredListeners is not initialized, leading to system crash

cnmicha, 2021-01-19:
The variable `m_nRegisteredListeners` (defined in https://github.com/mcci-catena/arduino-lorawan/blob/master/src/Arduino_LoRaWAN.h#L687) seems to be never initialized.

On my ESP32 platform, when sending data using `Arduino_LoRaWAN::SendBuffer`, this leads to the value being 4278124286. Then, it crashes.

> Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled. ([see ESP32 manual](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#instrfetchprohibited))

For me, the problem does not occur, if I change line 687 to `uint32_t m_nRegisteredListeners = 0;`.
Then, the LMIC starts joining and sending data, which works as expected.

Here are parts of my code:

```
class MyLoRaWAN : public Arduino_LoRaWAN_ttn {
public:
    MyLoRaWAN(){};

    static MyLoRaWAN *getInstance();

    bool sendBuffer(const uint8_t *pBuffer, size_t nBuffer, SendBufferCbFn *pDoneFn = nullptr, void *pCtx = nullptr,
                    bool fConfirmed = false);
}

static MyLoRaWAN *instance = nullptr;

MyLoRaWAN *MyLoRaWAN::getInstance() {
    if (instance == nullptr) {
        instance = new MyLoRaWAN();
    }

    return instance;
}

void setup() {
    MyLoRaWAN::getInstance()->begin();
}

void loop() {
    MyLoRaWAN::getInstance()->loop();
    // ... some application code triggering MyLoRaWAN::getInstance()->sendBuffer(...)
}
```

cnmicha, 2021-01-19:
The problem occurs because of the exit condition of the for loop (https://github.com/mcci-catena/arduino-lorawan/blob/master/src/lib/arduino_lorawan_begin.cpp#L122)

terrillmoore, 2021-01-19:
My understanding of C++ is that class members that are not explicitly initialized are zero at birth.

Please check your linker scripts, I think you have a problem in your system build.

cnmicha, 2021-01-19:
Thanks a lot.

I figured out the problem comes down to the constructor of the `MyLoRaWAN` class.

When defining the class as shown in the readme file, the crash occurs:

```cpp
class MyLoRaWAN : public Arduino_LoRaWAN_ttn {
public:
    MyLoRaWAN() {};
    // ...
```

Using a trivial default constructor (see [here](https://en.cppreference.com/w/cpp/language/default_constructor)), everything works:
```cpp
class MyLoRaWAN : public Arduino_LoRaWAN_ttn {
public:
    MyLoRaWAN() = default;
    // ...
```

Even declaring no constructor (it will implicitly declare a default constructor) works:
```cpp
class MyLoRaWAN : public Arduino_LoRaWAN_ttn {
public:
    // ...
```

closed 2021-01-19 by cnmicha