mcci-catena/arduino-lorawan #148: Requesting info about AS923 Japan support

kevin192291, 2021-02-28:
I have recently been looking into restrictions for LoRaWAN devices in Japan.
I found the following requirements:
```
Transmission time restrictions are:
  (1) If carrier sensing (LBT) function is provided: TX ON < 4s and TX OFF > 50ms
or
  (2) If carrier sensing (LBT) function is not provided: TX ON < 0.1s and TX OFF > 0.1ms, total TX time < 3.6s/hr
- Max. Antenna gain < 3dBi
- Max. Conducted power < 0.25W
- Other condition: device contained in sealed casing
```

My question is if the Arduino-lorawan library handles this when setting the frequency plan in the config(ARDUINO_LMIC_CFG_NETWORK_TTN) to AS923 Japan, or if this is something that I would be
required to handle myself?

Thank you very much for the info, and I am sorry if this question doesn't make sense.
I am still trying to learn all of this.

-Kevin

terrillmoore, 2021-02-28:
Hi, this is mostly handled by the LMIC. The LMIC implements LBT, so the restrictions are met. The "max antenna gain" is handled by your hardware design. Th max conducted power is handled by your hardware design; the LMIC attempts to tell the SX1276 what power to transmit at, but has no control over what the circuit actually does.

Since arduino-lorawan is really just a wrapper for [arduino-lmic](https://github.com/mcci-catena/arduino-lmic), you may want to open discussions there.

Best regards,
--Terry

kevin192291, 2021-02-28:
Thank you very much for the extremely fast reply. This helps a lot.

closed 2021-02-28 by kevin192291