mcci-catena/arduino-lorawan #151: Feature Request: Arduino Portenta LoRA Vision Shield

hpssjellis, 2021-04-11:
Any chance of making this library compatible with the [Arduino Portenta LoRA Vision Shield](https://store.arduino.cc/usa/portenta-vision-shield-lora)     

The machine learning capabilities of this shield make it potentially very useful on any LoRaWan network, but because of a hardware decision to send pins from the LoRa chip to the Murata module it is fairly difficult to have full access to LoRa so I am not sure if it will connect with this library or many other LoRa/LoRaWan libraries other than the MKRWAN library. Any opinions?

terrillmoore, 2021-04-11:
MCCI uses the Murata module all the time: as a CPU. However, you have to run the LMIC on the Murata module. In this case, the Murata is a peripheral to the H7. If you want to use the LMIC on the H7, you have to write a new radio driver that passes radio requests to the Murata module and gets back results; and then you have to add stubs on the Murata module to take the requests from the H7 and pass them to the SX1276. I'm afraid this is fairly complicated.

hpssjellis, 2021-04-12:
Thanks for the reply @terrillmoore you have clued me in to what is happening with some code I am using that has the Murata module working separately from the M7 and M4 cores.

closed 2021-05-08 by terrillmoore