mcci-catena/arduino-lorawan #159: Get ack using TTN V3

mhmayyan, 2021-05-11:
Using TTN V2 , usually, I use the flags combination LMIC.txrxFlags & TXRX_ACK inside the function Arduino_LoRaWAN::ReceivePortBufferCbFn receiveMessage. If that combination is true then I can tell that I got ack. Now, however, on TTN V3 it seems I don't receive a downlink at all.

Any idea how I can get ack and receive downlink messages using TTN V3?

Thanks,

terrillmoore, 2021-05-12:
This is because of issue #25. Try updating your arduino-lorawn to [v0.9.0-pre12](https://github.com/mcci-catena/arduino-lorawan/releases/tag/v0.9.0-pre12) -- you'll also have to update Catena-Arduino-Platform to [v0.21.0-pre5](https://github.com/mcci-catena/Catena-Arduino-Platform/releases/tag/v0.21.0-pre5). This will cause the Catena to properly save the 5-second downlink window setting that's communicated with a JoinAccept message.

Note that it will take 5 seconds to get a downlink, instead of 1.

mhmayyan, 2021-05-12:
Thank you, Terry

It worked. I had also to update arduino-lmic to the latest pre-release

terrillmoore, 2021-05-31:
Excellent. Closing issue.

closed 2021-12-10 by mhmayyan