mcci-catena/arduino-lorawan #165: Purpose of NetSaveSessionInfo

etix, 2021-06-02:
`NetSaveSessionInfo` is supposed to be used to store session info like `NwkSKey` and `AppSKey` in conjunction with the other callback `GetSavedSessionInfo` but the latter is never called. Is there another way to restore the `SessionInfo` data?

terrillmoore, 2021-06-02:
Thanks for your question.  (Which version are you commenting on? V0.9.0?)

In any case, it's imperfectly documented, but see this code. TL;DR: you have to implement `getAbpProvisioningInfo()` for saved sessions on OTAA devices.

https://github.com/mcci-catena/arduino-lorawan/blob/9b664f585941f30fbb69d833d9bbad9f59192ac5/src/lib/arduino_lorawan_begin.cpp#L51-L93

etix, 2021-06-02:
Sorry yes, I'm actually upgrading one of my node to 0.9.0 (previously in 0.8.x) in EU868 region.

I'll have a look to `getAbpProvisioningInfo()` and report back.

Thanks.

etix, 2021-06-02:
I've been able to make it work using `getAbpProvisioningInfo()`. 

For time travelers who end up on this issue, in addition to the obvious `GetOtaaProvisioningInfo()` I've also implemented:
- NetSaveSessionState
- NetGetSessionState
- NetSaveSessionInfo
- GetAbpProvisioningInfo

For the storage, I'm using the SerialFlash library with the onboard flash chip of my board (Mini Ultra Pro from Rocket Scream).

In `NetSaveSessionState()` I'm writing the whole `SessionState` struct into a file.
In `NetGetSessionState()` I'm doing the opposite, reading the file and writing back into the provided `SessionState` pointer.
In `NetSaveSessionInfo()` I'm writing the whole `SessionInfo` struct into another file (ignoring pExtraInfo and nExtraInfo).
In `GetAbpProvisioningInfo()` it's a bit more tricky since not everything is in the previously saved `SessionInfo`. For `FCntUp` and `FCntDown` I'm calling `NetGetSessionState()` from within `GetAbpProvisioningInfo()` to fetch the counters.

`GetSavedSessionInfo()` is effectively useless and should probably be removed to avoid confusion.

After a very quick test everything seems to be working (for EU868 at least), no more join request after a cold start.

Thanks again @terrillmoore for the quick answer.

closed 2021-06-02 by etix