mcci-catena/arduino-lorawan #181: Kernel panic on Arduino v2.0

olicooper, 2021-10-15:
The library worked fine on Arduino 1.0.6, but after upgrading to v2.0.0 the device reboots on the next loop cycle after calling `Arduino_LoRaWAN::SendBuffer()`. I have a `Lora` wrapper class which simply calls `Arduino_LoRaWAN::loop()` - it is failing at this point.

### environment:
* Version 0.9.1
* Ardunio v2.0.0 (I also tried the [`mem-optimized` branch](https://github.com/espressif/arduino-esp32/tree/77f1ebd59ef83c26427451446c0721d7742a9711))
* PlatformIO on Windows 10
* Heltec Wireless Stick v1.2

### code
```cpp
void Lora::NetBeginRegionInit() {
    this->Super::NetBeginRegionInit();
}

void Lora::NetJoin() {
    this->Super::NetJoin();
    SetLinkCheckMode(true);
    LMIC_setPingable(4);
}

void Lora::NetTxComplete() {
    this->Super::NetTxComplete();
}

bool Lora::GetOtaaProvisioningInfo(OtaaProvisioningInfo *pInfo) {
    if (pInfo != nullptr) {
        memcpy_P(pInfo->AppKey, APPKEY, sizeof(pInfo->AppKey));
        memcpy_P(pInfo->DevEUI, DEVEUI, sizeof(pInfo->DevEUI));
        memcpy_P(pInfo->AppEUI, APPEUI, sizeof(pInfo->AppEUI));
    }
    return true;
}

void Lora::loop() {
    if (!enabled) return;
    Super::loop(); // <-- Failing here
}
```

* My application flow is:
```cpp
// these are called in setup
begin(&pinMap);
os_setCallback(&basicStatePublishJob, processLMICCallback);

// ... then 'basicStatePublishJob' calls SendBuffer
SendBuffer(dataBuffer, dataBuffer_s, pDoneFn, (void *)this, /*confirmed*/ 0, /*port*/ 1);
```

### core dump
```log
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.

Core  1 register dump:
PC      : 0x400f4ed7  PS      : 0x00060030  A0      : 0x800f4f55  A1      : 0x3ffce000
A2      : 0x4002f56e  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x0013a16a  
A6      : 0xfffffffc  A7      : 0x00000003  A8      : 0x00000000  A9      : 0x3ffc7c18
A10     : 0x00000010  A11     : 0x00000016  A12     : 0x00000000  A13     : 0x00040df7  
A14     : 0x63ffffff  A15     : 0x00000003  SAR     : 0x0000001b  EXCCAUSE: 0x00000003
EXCVADDR: 0x4002f56e  LBEG    : 0x400f4e88  LEND    : 0x400f4ee6  LCOUNT  : 0x00000004  

0x400f4ed4: Arduino_LoRaWAN::BuildSessionState(Arduino_LoRaWAN::SessionState_u&) const at F:\myproject/.pio/libdeps/heltec_wireless_stick/MCCI Arduino LoRaWAN Library/src/Arduino_LoRaWAN.h:369
 (inlined by) ?? at F:\myproject/.pio/libdeps/heltec_wireless_stick/MCCI Arduino LoRaWAN Library/src/Arduino_LoRaWAN.h:359
 (inlined by) Arduino_LoRaWAN::BuildSessionState(Arduino_LoRaWAN::SessionState_u&) const at F:\myproject/.pio/libdeps/heltec_wireless_stick/MCCI Arduino LoRaWAN Library/src/lib/arduino_lorawan_sessionstate.cpp:114
0x400f4f52: Arduino_LoRaWAN::SaveSessionState() at F:\myproject/.pio/libdeps/heltec_wireless_stick/MCCI Arduino LoRaWAN Library/src/lib/arduino_lorawan_sessionstate.cpp:190
0x400f4b0b: Arduino_LoRaWAN::StandardEventProcessor(unsigned int) at F:\myproject/.pio/libdeps/heltec_wireless_stick/MCCI Arduino LoRaWAN Library/src/Arduino_LoRaWAN.h:925
 (inlined by) Arduino_LoRaWAN::StandardEventProcessor(unsigned int) at F:\myproject/.pio/libdeps/heltec_wireless_stick/MCCI Arduino LoRaWAN Library/src/lib/arduino_lorawan_begin.cpp:214
0x400f4be2: Arduino_LoRaWAN::DispatchEvent(unsigned int) at F:\myproject/.pio/libdeps/heltec_wireless_stick/MCCI Arduino LoRaWAN Library/src/lib/arduino_lorawan_begin.cpp:123
0x400f4c24: onEvent at F:\myproject/.pio/libdeps/heltec_wireless_stick/MCCI Arduino LoRaWAN Library/src/lib/arduino_lorawan_begin.cpp:109
0x400f07fd: reportEventNoUpdate at F:\myproject/.pio/libdeps/heltec_wireless_stick/MCCI LoRaWAN LMIC library/src/lmic/lmic.c:481
0x400f1d3d: reportEventAndUpdate at F:\myproject/.pio/libdeps/heltec_wireless_stick/MCCI LoRaWAN LMIC library/src/lmic/lmic.c:466
0x400f203a: startJoining at F:\myproject/.pio/libdeps/heltec_wireless_stick/MCCI LoRaWAN LMIC library/src/lmic/lmic.c:2142
0x400f308c: os_runloop_once at F:\myproject/.pio/libdeps/heltec_wireless_stick/MCCI LoRaWAN LMIC library/src/lmic/oslmic.c:157
0x400f4d6b: Arduino_LoRaWAN::loop() at F:\myproject/.pio/libdeps/heltec_wireless_stick/MCCI Arduino LoRaWAN Library/src/lib/arduino_lorawan_loop.cpp:42
0x400d5c03: Lora::loop() at F:\myproject/src/lora.cpp:156
0x400d680e: loop() at F:\myproject/src/main.cpp:196
0x400fb8c4: loopTask(void*) at C:/Users/Oliver/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:46
```

Related source code

https://github.com/mcci-catena/arduino-lorawan/blob/caceeb49b1f5520d11668aeb7c587ef1a96ca9ad/src/lib/arduino_lorawan_sessionstate.cpp#L190

https://github.com/mcci-catena/arduino-lorawan/blob/caceeb49b1f5520d11668aeb7c587ef1a96ca9ad/src/lib/arduino_lorawan_sessionstate.cpp#L114

https://github.com/mcci-catena/arduino-lorawan/blob/caceeb49b1f5520d11668aeb7c587ef1a96ca9ad/src/Arduino_LoRaWAN.h#L359-L371

terrillmoore, 2021-10-15:
We haven't tested on Arduino v2.0.0, and I unfortunately don't even use PlatformIO, though of course we support it -- it doesn't have support for our BSPs. And.. this is the first I've heard of Arduino 2.0. So I'm really out in the weeds and not able to invest much time short term.

You can possibly use `objdump` on the file `arduino_lorawan_sessionstate.`o to get the disassembly. But it depends on the tool chain. You'll have to research the command line switches on your version. `objdump` will be with `gcc`. There is an option to disassmble, and to include line numbers, and possibly even to include source code.  The `.o` file will be somewhere in your build directory.

terrillmoore, 2021-10-17:
@olicooper email is fine, but it hasn't come yet. Checked spam filters also, nothing there.

terrillmoore, 2021-10-17:
can you send the corresponding link map?

terrillmoore, 2021-10-17:
The problem is not what I thought. The code for doing the store is correct. Removing the debug `.loc` stuff:

```
# .pio/libdeps/heltec_wireless_stick/MCCI Arduino LoRaWAN Library/src/Arduino_LoRaWAN.h:368:                         chPtr[0] = uint8_t(reducedFreq >> 16);
	s8i	a4, a2, 0	# *chPtr_113, tmp200
# .pio/libdeps/heltec_wireless_stick/MCCI Arduino LoRaWAN Library/src/Arduino_LoRaWAN.h:369:                         chPtr[1] = uint8_t(reducedFreq >> 8);
	s8i	a12, a2, 1	# MEM[(unsigned char *)chPtr_113 + 1B], tmp201
# .pio/libdeps/heltec_wireless_stick/MCCI Arduino LoRaWAN Library/src/Arduino_LoRaWAN.h:370:                         chPtr[2] = uint8_t(reducedFreq);
	s8i	a3, a2, 2	#, reducedFreq
```

Those are three byte writes to the pointer. So I need the link map to find exactly where things are failing.

But... could this be related to initialization and switching back and forth (described in other report)? The LMIC does not do much checking for invalid state, and since the compiled code is correct, my guess is that some preconditions we are depending on are not true.

olicooper, 2021-10-17:
In my code I read from NVS storage during boot to decide if I should start the lora 'module' or not. If I have enabled the module it will start normally by calling the required `Arduino_LoRaWAN::begin(&pinMap);` function, then calling `loop()` as normal. There is no runtime switching happening. I should note that the code works fine with Arduino 1.0.6. 
I have sent you an updated email with all output. Let me know if there is anything else you need me to do. Thank you!

olicooper, 2021-10-20:
I mentioned on the other ticket that this should be the correct ELF, but I will recompile the elf again against the Arduino `main` branch to be sure.
You are correct that you'd need a LoRa module to execute this code which does make it more challenging.
The odd thing about it is that I can compile the exact same code on Arduino v1.0.6 and it works perfectly. Switching to v2.0.0 (and subsequently IDF 4.3) cases the crash.

terrillmoore, 2022-05-03:
The assert issue is due to calling `radio_rand1()` before the buffer is initialized. Not sure why that might be happening.