mcci-catena/arduino-lorawan #202: Print Spread Factor, and Power Consumption

jndqadar, 2022-07-28:
Dear Mr. Terry Moore (@terrillmoore ),

I am using LMIC code for experiment purposes and would like to print the spread factor and power consumed for each payload sent to the Server. I do wanna look at both parameters i,e SF and Power consumption on Serial console on Arduino IDE. Would you please provide me with your kind instruction on how to do that? 

`LMIC_setLinkCheckMode(0);`
`LMIC_setDrTxpow(DR_SF7,14);`

I would like to change the value from 14 dBi to 20 dBi and DR_SF7 to DR_SF12 and do wanna see the result on Arduino Console.

Looking forward to hearing from you
Thank you

Best,

[image]

terrillmoore, 2022-07-28:
Hi, sure. My recommendation is to capture this info during EV_TXSTART processing, and then print it later when it's not time critical (e.g., during EV_JOIN_TXCOMPLETE).  (Printing takes time and the LMIC has hard-real-time paths.)

With respect to Tx power, there is some tricky handoff between the LMIC and the radio driver, I'll have to look up the exact correct variable to capture. Which version of LMIC are you using?

jndqadar, 2022-07-28:
> EV_TXSTART

Dear Mr. Terry Moore,

Thanks for your prompt response. What I understood from your recommendation is to capture SPREAD_FACTOR information in the main code (**ttn-otaa-feather-us915-dht22.ino**) file, or do we have to define it in the **LMIC.C** file? 

      `case EV_TXSTART:
            Serial.println(F("EV_TXSTART"));
            break;
        case SPREAD_FACTOR:      // <=================
            break;
        case EV_TXCANCELED:
            Serial.println(F("EV_TXCANCELED"));
            break;
        case EV_RXSTART:
            /* do not print anything -- it wrecks timing */
            break;
        case EV_JOIN_TXCOMPLETE:
            Serial.println(F("SPREAD_FACTOR: Uplink Packet sent with Spread Factor"));    // <=================
            Serial.println(F("EV_JOIN_TXCOMPLETE: no JoinAccept"));
            break;` 

I added those two lines but it gives me an error that "SPREAD_FACTOR" was not declared in this scope. For sure it must show this error because we didn't define it in the code. Would you be kind enough to please let me know where and how can we define the statement for the Spread Factor? 

______________
For **Powe consumption**, I kindly looking forward to hearing from you, when you get a chance in your free time. I tried myself but couldn't solve the issue. 

I am using the latest version of **LMIC [4.1.1]**

Thank you

terrillmoore, 2022-07-28:
Sorry, I was not clear. While processing the event `EV_TXSTART`, the code should cache the values of:

* `LMIC.rps`: this has the spreading factor and other information.  To get the spreading factor, you use `getSf(rps)` from `lorabase.h`.
*  
* `LMIC.radio_txpow`: this has the requested transmit power, in dB.

Please refer to doc/RadioDriver.md for more info.

terrillmoore, 2022-07-29:
As this isn't really a bug, converting to a discussion.