mcci-catena/arduino-lorawan #218: I get much errors durring compiling in plattform io

martinmmi, 2023-05-05:
Hello :)

Iam martin from germany and i try to connect my ttgo v21 boards with the the things network. Unfortunately I get a lot of errors in platform io so I don't know how to start. :(

**Here are my steps so far:** 
1. New Project in plattform io
2. Clone the "arduino_lorawan_esp32_example.ino" into the new projekt
3. Install the following libs from the example via the internal plattform io lib installer:

MCCI Arduino Development Kit ADK     0.2.2
MCCI LoRaWAN LMIC library                 4.1.1
MCCI Arduino LoRaWAN Library           0.9.2
LoRa_Serialization                                   3.2.1

4. Added the files secrets.h and LoraMessage.h according to the description in the source code / git.
5. Try to compile. 

**Here a screenshot from the errors:**
[image]

**Here a screenshot from the secrets.h and loramessage.h:**
[image]

[image]

Can you give me a small support via discord?

martinmmi, 2023-05-05:
Here is the repo, what i opened for this project:
https://github.com/martinmmi/ttn_mqtt

matthias-bs, 2023-05-05:
Well, I'm afraid I cannot provide this kind of support!
And normally I reject any questions concerning PlatformIO, because I'm only using the Arduino IDE - but today is your lucky day!

Please remove your [src/LoraMessage.h](https://github.com/martinmmi/ttn_mqtt/blob/main/src/LoraMessage.h). This hides a file with the same name provided by the library LoraSerialization. The correct board definition is provided by some magic from `platformio.ini`.

Then change `platformio.ini` as follows:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:ttgo-lora32-v21]
platform = espressif32
board = ttgo-lora32-v21
framework = arduino
lib_deps = 
	mcci-catena/MCCI LoRaWAN LMIC library@^4.1.1
	mcci-catena/MCCI Arduino Development Kit ADK@^0.2.2
	mcci-catena/MCCI Arduino LoRaWAN Library@^0.9.2
	thesolarnomad/LoRa Serialization@^3.2.1
	SPI

build_flags =
    ; Use platformio.ini for settings instead lmic_project_config.h.
    -D ARDUINO_LMIC_PROJECT_CONFIG_H_SUPPRESS

    ; Ping and beacons not supported for class A, disable to save memory.
    -D DISABLE_PING
    -D DISABLE_BEACONS
    -D hal_init=LMICHAL_init           ; Workaround for naming conflict of function hal_init
                                       ; introduced by newer versions (> 3.5.0) of 
                                       ; PlatformIO Espressif 32 platform (ESP32 Arduino core).
                                       ; See https://github.com/lnlp/LMIC-node/issues/41 for more information.

    ; -D LMIC_DEBUG_LEVEL=1            ; 0, 1 or 2
    
    ; -D CFG_sx1272_radio=1            ; Use for SX1272 radio
    -D CFG_sx1276_radio=1              ; Use for SX1276 radio
    -D USE_ORIGINAL_AES                ; Faster but larger, see docs
    ; -D LMIC_USE_INTERRUPTS           ; Not tested or supported on many platforms
    -D LMIC_ENABLE_DeviceTimeReq=1   ; Network time support

    ; --- Regional settings -----
    ; Enable only one of the following regions:    
    ; -D CFG_as923=1
    ; -D CFG_as923jp=1   
    ; -D CFG_au915=1
    ; -D CFG_cn490=1                   ; Not yet supported
    ; -D CFG_cn783=1                   ; Not yet supported
    ; -D CFG_eu433=1                   ; Not yet supported
    -D CFG_eu868=1
    ; -D CFG_in866=1
    ; -D CFG_kr920=1
    ; -D CFG_us915=1
```
The first thing was that the SPI library was missing - it's built in with the Arduino IDE, but here it has to be provided explicitely.

Then we have some LMIC specific stuff - parts of it more or less easy to understand (defines for the radio module and the LoRaWAN region), other parts of it... not so obvious (proudly copied from here: https://github.com/lnlp/LMIC-node/blob/main/platformio.ini).

Finally we have two warnings which have to be fixed:
1. Warning: `main.cpp`: `no return statement in function returning non-void [-Wreturn-type]`
   Fix: please change
   ```
   bool uplinkRequest(void) {
        m_fUplinkRequest = true;
   };
   ```
   to
   ```
   void uplinkRequest(void) {
        m_fUplinkRequest = true;
   };
  ````
2. Warning: `ttn_mqtt/.pio/libdeps/ttgo-lora32-v21/MCCI Arduino LoRaWAN Library/src/Arduino_LoRaWAN.h`: 
	`control reaches end of non-void function [-Wreturn-type]` [Ln370, Col25]
   Please see [mcci-catena/arduino-lorawan/pull/204](https://github.com/mcci-catena/arduino-lorawan/pull/204)

Good luck!

matthias-bs, 2023-05-05:
BTW: You can safely ignore this warning: `#warning "You must replace the values marked FILLMEIN with real values from the TTN control panel!" [-Wcpp]`

martinmmi, 2023-05-08:
Omg. Thank you so much! Seems like it's working!!! 
[image]

martinmmi, 2023-05-08:
One more question: 

How can I connect to the follow gateway id (its only 500m left from my home): **Gateway ID: oi-lej-gw01**
How can i say that my device connect with this gateway? Or will it do this automatically?

And I have much authenification id's on ttn. Which one i have to use for **appeui** in the abp mode?

deveui[] = { 0x3D, 0xD4, 0x05, 0xD0, 0x7E, 0xD5, 0xB3, 0x70 };                LITTLE-ENDIAN
appkey[] = { 0x20, 0xD9, 0xAB, 0x7A, 0xD7, 0x60, 0x4A, 0xC5, 0x6E, 0x6E, 0x3B, 0x35, 0x38, 0x92, 0x3F, 0xAB };                  BIG-ENDIAN

appeui[] = NwkSKey ???

[image]

matthias-bs, 2023-05-08:
Hi Martin,

I don't know if you can restrict the connection to a specific gateway. AFAIK, if a node is received by multiple gateways the network handles it in the proper way (e.g. avoiding transmission of duplicate payload messages)

Regarding the IDs, have a look at https://www.thethingsindustries.com/docs/reference/glossary/ or in the TTN console, select your device, go to the tab "General settings" and hover over the help in the sections "Network layer" and "Application layer". **The "S" in some of the IDs stands for "secret" - you should not expose those to the rest of the world...**

Normally (unless to have hardware/software restrictions which prevent this, which is not the case with ESP32 and LMIC-based SW)), you should use OTAA instead of APB. You can find an explanation why on the excellent TTN website as well.

Cheers
Matthias

matthias-bs, 2023-05-08:
BTW: You can at least use LoRaWAN specification 1.0.3. The regional parameters are selected automatically in this case. Not sure if a later version would also work...

martinmmi, 2023-05-09:
Okay, thank you so much. It works! I coulded connect with the gateway oi-lej-gw01.

13:42:52 Forward join-accept message
13:42:51 Accept join-request

Is the example based on abp mode, or?

https://github.com/mcci-catena/arduino-lorawan/blob/master/examples/arduino_lorawan_esp32_example/arduino_lorawan_esp32_example.ino

When i like to try the otaa mode, i have to use this example, maybe? 
https://github.com/mcci-catena/arduino-lmic/tree/master/examples/ttn-otaa

Thanks so much!

matthias-bs, 2023-05-09:
The ESP32 example uses. OTAA.

kadirozdinc, 2023-05-14:
I try to make the sketch work but I got an error. I use ESP32 DOIT V1 dev board and HopeRF RFM96 868mhz Radio module. I made those changes mentioned above I could compile successfully but esp32 restarts itself continously. 

My repo : https://github.com/kadirozdinc/esp32-LorawanLMIC

Error in Serial Console : 

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d561d  PS      : 0x00060230  A0      : 0x800d56c0  A1      : 0x3ffb2060  
A2      : 0x3ffb20ec  A3      : 0x00847628  A4      : 0x00000084  A5      : 0x3ffc35cc
A6      : 0x00008476  A7      : 0x3ffb20cc  A8      : 0x3ffc35cc  A9      : 0x33be27a0
A10     : 0x3ffb20cc  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffc35cc
A14     : 0x3ffb211c  A15     : 0x63ffffff  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00847658  LBEG    : 0x400864f8  LEND    : 0x40086503  LCOUNT  : 0x00000000

Backtrace: 0x400d561a:0x3ffb2060 0x400d56bd:0x3ffb2090 0x400d50f7:0x3ffb2190 0x400d5190:0x3ffb21b0 0x400d51d0:0x3ffb21d0 0x400d19d5:0x3ffb21f0 0x400d2788:0x3ffb2210 0x400d38fd:0x3ffb2230 0x400d550f:0x3ffb2250 0x400d1954:0x3ffb2270 0x400d7021:0x3ffb2290

ELF file SHA256: ca8aa52e3b79a85a

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
```