mcci-catena/arduino-lorawan #219: Rebooting issue in esp32 example

kadirozdinc, 2023-05-14:
I use ESP32 DOIT V1 dev board and HopeRF RFM96 radio module. I add required library,  edit platformio.ini file and could compile the sketch successfully. When upload code into dev board, device reboots itseflt continously like : 

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d561d  PS      : 0x00060230  A0      : 0x800d56c0  A1      : 0x3ffb2060
A2      : 0x3ffb20ec  A3      : 0x00847628  A4      : 0x00000084  A5      : 0x3ffc35cc  
A6      : 0x00008476  A7      : 0x3ffb20cc  A8      : 0x3ffc35cc  A9      : 0x33be27a0
A10     : 0x3ffb20cc  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffc35cc  
A14     : 0x3ffb211c  A15     : 0x63ffffff  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00847658  LBEG    : 0x400864f8  LEND    : 0x40086503  LCOUNT  : 0x00000000  

Backtrace: 0x400d561a:0x3ffb2060 0x400d56bd:0x3ffb2090 0x400d50f7:0x3ffb2190 0x400d5190:0x3ffb21b0 0x400d51d0:0x3ffb21d0 0x400d19d5:0x3ffb21f0 0x400d2788:0x3ffb2210 0x400d38fd:0x3ffb2230 0x400d550f:0x3ffb2250 0x400d1954:0x3ffb2270 0x400d7021:0x3ffb2290

ELF file SHA256: ca8aa52e3b79a85a

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d561d  PS      : 0x00060230  A0      : 0x800d56c0  A1      : 0x3ffb2060
A2      : 0x3ffb20ec  A3      : 0x00847628  A4      : 0x00000084  A5      : 0x3ffc35cc  
A6      : 0x00008476  A7      : 0x3ffb20cc  A8      : 0x3ffc35cc  A9      : 0x33be27a0
A10     : 0x3ffb20cc  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffc35cc  
A14     : 0x3ffb211c  A15     : 0x63ffffff  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00847658  LBEG    : 0x400864f8  LEND    : 0x40086503  LCOUNT  : 0x00000000  

Backtrace: 0x400d561a:0x3ffb2060 0x400d56bd:0x3ffb2090 0x400d50f7:0x3ffb2190 0x400d5190:0x3ffb21b0 0x400d51d0:0x3ffb21d0 0x400d19d5:0x3ffb21f0 0x400d2788:0x3ffb2210 0x400d38fd:0x3ffb2230 0x400d550f:0x3ffb2250 0x400d1954:0x3ffb2270 0x400d7021:0x3ffb2290

ELF file SHA256: ca8aa52e3b79a85a

Rebooting...
```
all connections I made  are : 

SCK - 18
MISO - 19
MOSI - 23
NSS      5
RST      13
DIO0     12
DIO1     14
DIO2     26

My Repo : https://github.com/kadirozdinc/esp32-LorawanLMIC

where do i make a mistake ?

matthias-bs, 2023-05-14:
Hi,
Could you please provide some information about the board (e.g . manufacturer's website, schematic or at least a pinout) and the compiler's log file?
Regards
Matthias

matthias-bs, 2023-05-15:
Hi,

I think most likely you just have to apply this patch:
https://github.com/mcci-catena/arduino-lorawan/pull/204

The wiring seems to be fine.

Best regards
Matthias

matthias-bs, 2023-05-15:
Please also apply the following fix in `main.cpp`:

Change
```
bool uplinkRequest(void) {
     m_fUplinkRequest = true;
};
```

to

```
void uplinkRequest(void) {
     m_fUplinkRequest = true;
};
```

kadirozdinc, 2023-05-15:
Thanks a lot, It worked for me. Another problem I faced is : 

I am monitoring current consumption of radio module with PowerProfilerKit2, After sleeping, current consumption is floating around 100uA. It seems radio cannot sleep successfully after sleeping with ESP32. but normally in my previous tests with Atmega328p, it was consuming around 6.5 uA after sleeping all the time.

Also If I dont use deepSleep, while not transmitting, current consumption  is constantly around 6.5uA. I wonder why consumption increases after sleeping ? Any idea ?

matthias-bs, 2023-05-15:
My guess is that the cause is due to an I/O leakage current. Is it just the RFM96 radio module current or the combined radio module/ESP32 current?

Maybe it helps to add a high-impedance pull-up (or pull-down) resistor to some I/O pins to achieve a defined level instead of a floating input.

kadirozdinc, 2023-05-15:
It is only radio module consumption. When I get closer my hand/body to circuit, current goes up or down,that is, it is affected by environment. There seems to be something floating after sleeping. I will examine that later on it.

Other than that I am also trying to implement downlink in the sketch. I looked over your repo : https://github.com/matthias-bs/BresserWeatherSensorTTN/blob/main/BresserWeatherSensorTTN.ino
I examine how you implement downlink and try to add that feature into my sketch like : 

I add the code below under Lorawan methods

```
// Receive and process downlink messages
void ReceiveCb(
    void *pCtx,
    uint8_t uPort,
    const uint8_t *pBuffer,
    size_t nBuffer) {
    DEBUG_PRINTF("--- Downlink Data ---\n");        
    DEBUG_PRINTF("Port: %d", uPort);
    char buf[255];
    *buf = '\0';

    if (uPort >= 0) {
        for (int i = 0; i < nBuffer; i++) {
              sprintf(&buf[strlen(buf)], "%02X ", pBuffer[i]);
        }
        DEBUG_PRINTF("Data: %s", buf);

    }
}
```
and add

`this->SetReceiveBufferBufferCb(ReceiveCb);` in cMyLoraWAN::setup function. 

I use chirpstack, in Queue tab, I enqueue some mesages to downlink, in sketch code branches into ReceiveCb function all the time every time mcu wakes up and messages doesn't disappear in chirpstack queue page after downlink. (should be dissapeared) Where I make a mistake ?

matthias-bs, 2023-05-15:
I don't know anything about chirpstack. Maybe you can find some hints here: https://github.com/matthias-bs/BresserWeatherSensorTTN#remote-configuration-via-lorawan-downlink

One question is how to provide downlink data - hex format, base64 or plain text?

My solution is not completed. I can receive data, but sending an uplink message after receiving a downlink did not work yet. I think the reason is that the sketch enters sleep mode to early. I did not find the time to work on this yet.

You could also have a look at this project: https://github.com/rgot-org/TheThingsNetwork_esp32
It seems to provide a higher level of abstraction. I have no experience with it, but from a first glance it looks promising.

kadirozdinc, 2023-05-15:
Thank you so much for your interest, I will try to implement downlinks. If you can make it let me know please :)

MajorTwip, 2023-05-27:
Should be closed by #204

PureTek-Innovations, 2023-08-16:
I have a similar and probably related issue with this sketch
I am using a TTGO LoRa module version 2.1 (1.61) , which I have selected in the Arduino IDE
I am trying to send data to a Chirpstack gateway so I have defined:

#define ARDUINO_LMIC_CFG_NETWORK_CHIRPSTACK 1
#define ARDUINO_LMIC_CFG_NETWORK_GENERIC 0

I have added APPEUI, DEVEUI and APPKEY in secrets.h

Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d198d  PS      : 0x00060830  A0      : 0x800d798d  A1      : 0x3ffb2260  
A2      : 0x00000001  A3      : 0x020035e6  A4      : 0x00000014  A5      : 0x00000004  
A6      : 0x3ffb8188  A7      : 0x80000001  A8      : 0x800d1984  A9      : 0x3ffb21c0  
A10     : 0x00000019  A11     : 0x9b5decb8  A12     : 0x9b5decb8  A13     : 0x3ffb21fc  
A14     : 0x3ffc29e0  A15     : 0x3ffb21fc  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000031  LBEG    : 0x40086969  LEND    : 0x40086979  LCOUNT  : 0xffffffff  

Backtrace: 0x400d198a:0x3ffb2260 0x400d798a:0x3ffb2290

ELF file SHA256: 63e631795b5d43e2

Which decodes to:

PC: 0x400d198d
EXCVADDR: 0x00000031

Decoding stack results
0x400d198a: printSessionInfo(Arduino_LoRaWAN::SessionInfo_u const&) at C:\Users\Jem\Documents\Arduino\LoRa\lorawan\arduino_lorawan_esp32_example\arduino_lorawan_esp32_example/arduino_lorawan_esp32_example.ino line 542
0x400d798a: loopTask(void*) at C:\Users\Jem\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9\cores\esp32\main.cpp line 42

With debugging disabled, the sketch still boot loops with the following errors:

load:0x40080400,len:3028
entry 0x400805e4
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d17e4: 4282180c 0000000e 32008136
Core  1 register dump:
PC      : 0x400d17e9  PS      : 0x00060830  A0      : 0x800d7395  A1      : 0x3ffb2260  
A2      : 0x3ffc257c  A3      : 0x00000000  A4      : 0x00000014  A5      : 0x00000004  
A6      : 0x3ffb8188  A7      : 0x80000001  A8      : 0x00000001  A9      : 0x3ffb2240  
A10     : 0x3ffc2594  A11     : 0x0001c200  A12     : 0x00000078  A13     : 0x00000003  
A14     : 0x00000001  A15     : 0x0000e100  SAR     : 0x00000000  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400d27c4  LEND    : 0x400d27d3  LCOUNT  : 0x00000000  

Backtrace: 0x400d17e6:0x3ffb2260 0x400d7392:0x3ffb2290

ELF file SHA256: f1a9ea6249950860

Which decodes to:

PC: 0x400d17e9
EXCVADDR: 0x00000000

Decoding stack results
0x400d17e6: setup() at C:\Users\Jem\Documents\Arduino\LoRa\lorawan\arduino_lorawan_esp32_example\arduino_lorawan_esp32_example/arduino_lorawan_esp32_example.ino line 423
0x400d7392: loopTask(void*) at C:\Users\Jem\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9\cores\esp32\main.cpp line 42

PureTek-Innovations, 2023-08-16:
Hi @terrillmoore 
name=MCCI Arduino LoRaWAN Library
version=0.9.2
Thank you

terrillmoore, 2023-08-16:
Please try updating to the 0.10 pre-release -- it's tagged. It has several bug fixes

PureTek-Innovations, 2023-08-17:
Thanks @terrillmoore
That seems to have worked on platformio on VSC.
Now I just need to get the Gateway to see it, I'm not convinced that it's configured properly...
I do have this error:
543 ms: NetGetSessionState() - failed
And these are blank
NetID:          0x00000000
DevAddr:        0x00000000
NwkSKey:        00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
AppSKey:        00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

matthias-bs, 2023-08-24:
I don't know if it is still of interest, bit I recently implemented downlink and uplinks triggered by downlink commands in my project:
https://github.com/matthias-bs/BresserWeatherSensorTTN

At least it is working with TTN and I provided some payload formatters.